hex-five/multizone-sdk #23: Documentation: reference toolchain and openocd

cgarlati, 2019-02-14:
Add documentation for using reference toolchain and openocd as submodules have been removed from this repo.

donbarnetson, 2019-02-17:
Updated manual is available in the multizone-sdk-manual repo; pdf is here:
https://github.com/hex-five/multizone-sdk-manual/blob/master/release/multizone_sdk_manual.pdf

donbarnetson, 2019-02-17:
Updated link

https://github.com/hex-five/multizone-sdk-manual/blob/master/multizone-sdk/release/multizone_sdk_manual.pdf

cgarlati, 2019-03-08:
Need to update again as the tarballs have been removed from MultiZone releases for licensing reason.

cgarlati, 2019-03-11:
https://hex-five.com/developers/ probably a good candidate for:

- reference riscv-gnu-toolchain
- reference riscv-openocd
- reference x300 mcs

closed 2020-01-11 by cgarlati