hex-five/multizone-sdk #34: InterZone overwrites messages

borancar, 2019-03-08:
### Description

InterZone seems to overwrite the posted message before it was received under certain circumstances.

Reported on: https://github.com/hex-five/multizone-sdk/commit/ba86fd33c96b9385922cd852f7c4ca3a5233d056
Bitstream: E31 v3p0

### Steps to reproduce

1. Clone or apply https://github.com/warpBytes/multizone-sdk/commit/317e659db1199c6c56104836b2960e7cdd56161e
2. Connect UART
3. Compile and upload

### Expected results

```
Sent message to zone 2
```

### Actual results

```
Sent message to zone 2
Sent message to zone 2
```

borancar, 2019-03-09:
I've replaced the printf in zone2 (receiver zone) with a load that will cause a trap and the issue still persists - https://github.com/warpBytes/multizone-sdk/commit/9f5e1f1a7cf83b935a452a0bee08fcdc918e427e

Talking with Cesare, seems that the 2 things need to happen - a preemption (or multiple) and a trap in the receiver zone to trigger this issue.

borancar, 2019-03-09:
Further simplifications done in https://github.com/warpBytes/multizone-sdk/commit/e5b224add0075c0c517fe6a85ea3c17476ebcfc5:
- Use volatile for loop variable instead of modifying Makefiles
- Put the wait loop directly inside main rather than a function call

cgarlati, 2019-03-09:
Fixed in cd441ae
hexfive-kern 5225687
hexfive-conf 0e3df93

closed 2019-03-09 by cgarlati