hex-five/multizone-sdk #37: PLIC (BTN0) broken in E31/S51 v19_02_rc0.mcs

cgarlati, 2019-03-13:
@borancar Check SiFive's latest documentation and freedom-e-sdk PLIC driver to see what may have changed.

borancar, 2019-03-15:
GPIO interrupts have changed in those bitstreams. The documentation link is broken and points to an older 3p0.

In this new bitstream, GPIO BTN_0 is at offset 5 rather than 11 in the PLIC:
https://github.com/sifive/freedom-e-sdk/blob/2983888c432211fe3eafa18fcf58c69f69473e9e/bsp/coreip-e31-arty/metal.h#L272-L282

The following change is needed to make it work:
```
diff --git a/zone2/main.c b/zone2/main.c
index cdac222..5c64c87 100644
--- a/zone2/main.c
+++ b/zone2/main.c
@@ -12,7 +12,7 @@
 #define LOCAL_INT_BTN_1  5
 #define LOCAL_INT_BTN_2  6
 
-#define GPIO_INT_BASE 7
+#define GPIO_INT_BASE 1
 #define INT_DEVICE_BUTTON_0 (GPIO_INT_BASE + BTN0)
 #define INT_DEVICE_BUTTON_1 (GPIO_INT_BASE + BTN1)
 #define INT_DEVICE_BUTTON_2 (GPIO_INT_BASE + BTN2)
```

Doing this in a portable way will take me some time (trying to find how to dynamically establish versions now).

borancar, 2019-03-15:
New bitstreams have the mimpid set to 0x20190228, older have it set to 0x00000000.

borancar, 2019-03-15:
Fixed in https://github.com/hex-five/multizone-sdk/pull/39.

closed 2019-03-27 by cgarlati