hex-five/multizone-sdk #43: Changing LD0 color doesn't work on S51

borancar, 2019-03-24:
Reported on: 0e5d4a83e0ecf851d920b6c4069bb02f6c19073d
Reproduced on boards: S51

### Description

Sending messages to zone3 to change the color has no effect.

### Steps to reproduce

```
Z1 > send 3 red

Z1 > send 3 gree

Z1 > send 3 blue
```

borancar, 2019-03-24:
This could be related to https://github.com/hex-five/multizone-sdk/issues/41.

borancar, 2019-04-09:
Narrowed down to an issue in the nanokernel (or documentation) - https://github.com/hex-five/multizone-sdk/tree/bug/send64. UART should print abcd but instead it prints ac.

cgarlati, 2019-05-25:
Fixed in [d508f88](https://github.com/hex-five/multizone-sdk/commit/d508f889be9654233fd66ac45e8fa6d6ceda989d)

closed 2019-05-25 by cgarlati