hex-five/multizone-sdk #48: make load failed

icennaise, 2020-09-27:
I have already burn X300 softcore on to Arty-35T and refence  [https://static.dev.sifive.com/SiFive-E310-arty-gettingstarted-v1.0.6.pdf](url) connect Arty-35T,host machine and ARM-USB-TINY-H, but when i try to run `make load`,an error occureed, it said:

/home/a/riscv-openocd-20200613/bin/openocd -f bsp/X300/openocd.cfg & \
/home/a/riscv-gnu-toolchain-20200613/bin/riscv64-unknown-elf-gdb multizone.hex --batch -ex "set mem inaccessible-by-default off" -ex "set remotetimeout 240" -ex "set arch riscv:rv32" -ex "target extended-remote localhost:3333" -ex "monitor reset halt" -ex "monitor flash protect 0 64 last off" -ex "load" -ex "monitor resume" -ex "monitor shutdown" -ex "quit"
Open On-Chip Debugger 0.10.0+dev-00860-g97fb3f4bd (2020-06-13-20:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
The target architecture is assumed to be riscv:rv32
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Info : Listening on port 3333 for gdb connections
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
openocd: src/target/riscv/riscv-011.c:431: dbus_scan: Assertion `info->addrbits != 0' failed.
localhost:3333:connect time out.
"monitor" command not supported by this target.
"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.
"monitor" command not supported by this target.

Is that means i connect wrong wires?
Thank you.

icennaise, 2020-09-28:
I have sloved this problem. I confirm that i connect wrong wires. It is that because my ARM-USB-TINY-H is little different from the page i refence: it has a additional wires and the first pin's location is different from which without that wires.
You can refence this page :
 [https://www.olimex.com/Products/ARM/JTAG/_resources/ARM-USB-TINY_and_TINY_H_manual.pdf](url)
It say:

**There is a small mark over the connector that indicates where the first pin is located. It might be
difficult to spot it at first glance (because of the casing) so please also consider that the cable that
comes with the debugger has the first wire colored in red.**

cgarlati, 2020-09-29:
Note for future readers:

Instructions on how to to connect the Olimex debug head ARM-USB-TINY-H to the ARTY board
[Arty FPGA Dev Kit Getting Started Guide](https://static.dev.sifive.com/SiFive-E310-arty-gettingstarted-v1.0.6.pdf)

closed 2020-09-29 by cgarlati

Aldwin97two, 2020-12-11:
Hello,

I have exactly the same problem and I've already checked the wires several times.

> home/aldwin/Documents/M2_CSSE/Project_HexFive/hexfive/softwares/riscv-openocd-20200613/bin/openocd -f bsp/X300/openocd.cfg & \
/home/aldwin/Documents/M2_CSSE/Project_HexFive/hexfive/softwares/riscv-gnu-toolchain-20200613/bin/riscv64-unknown-elf-gdb multizone.hex --batch -ex "set mem inaccessible-by-default off" -ex "set remotetimeout 240" -ex "set arch riscv:rv32" -ex "target extended-remote localhost:3333" -ex "monitor reset halt" -ex "monitor flash protect 0 64 last off" -ex "load" -ex "monitor resume" -ex "monitor shutdown" -ex "quit"
Open On-Chip Debugger 0.10.0+dev-00860-g97fb3f4bd (2020-06-13-20:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
The target architecture is assumed to be riscv:rv32
Info : Listening on port 3333 for gdb connections
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Error: Unsupported DTM version: 15
Remote communication error.  Target disconnected.: Connexion ré-initialisée par le correspondant.
"monitor" command not supported by this target.
"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.
"monitor" command not supported by this target.

[image]

[image]

Is it properly plugged in? 
Is there another cause for this problem? 
Thank you.