hex-five/multizone-sdk #49: failed to upload the .hex file to the Arty 100T

Aldwin97two, 2020-12-14:
Hello,

I try to redo the demonstration of Multizone secure Iot Stack but arrived at the step to upload the .hex file to the Arty 100T card and the debbuger JTAG ARM-USB-TINY-H, I have the following error :

> home/aldwin/Documents/M2_CSSE/Project_HexFive/hexfive/softwares/riscv-openocd-20200613/bin/openocd -f bsp/X300/openocd.cfg & \
/home/aldwin/Documents/M2_CSSE/Project_HexFive/hexfive/softwares/riscv-gnu-toolchain-20200613/bin/riscv64-unknown-elf-gdb multizone.hex --batch -ex "set mem inaccessible-by-default off" -ex "set remotetimeout 240" -ex "set arch riscv:rv32" -ex "target extended-remote localhost:3333" -ex "monitor reset halt" -ex "monitor flash protect 0 64 last off" -ex "load" -ex "monitor resume" -ex "monitor shutdown" -ex "quit"
Open On-Chip Debugger 0.10.0+dev-00860-g97fb3f4bd (2020-06-13-20:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
The target architecture is assumed to be riscv:rv32
Info : Listening on port 3333 for gdb connections
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Error: Unsupported DTM version: 15
Remote communication error.  Target disconnected.: Connexion ré-initialisée par le correspondant.
"monitor" command not supported by this target.
"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.
"monitor" command not supported by this target.

I've already checked the wires several times.

[image]

[image]

Is it properly plugged in? 
Is there another cause for this problem? 
Thank you.

cgarlati, 2020-12-14:
What bitstream did you upload to the FPGA?

[Readme.md:](https://github.com/hex-five/multizone-sdk/blob/master/README.md)

The Xilinx Arty FPGA comes in two versions: 35T and 100T

    Hex Five's X300 bitstream works with version 35T
    SiFive's bitstreams up to v19.02 work with version 35T
    SiFive's bitstreams after v19.02 work with version 100T

Aldwin97two, 2020-12-14:
I upload the **X300ArtyDevKitFPGAChip-1ab2531.mcs** file from this link https://github.com/hex-five/multizone-fpga/releases

cgarlati, 2020-12-14:
Your hardware (100T) is not compatible with the X300 bitstream.

The Hex Five's X300 bitstream requires the Xilinx Arty FPGA 35T.

closed 2020-12-14 by cgarlati