hex-five/multizone-sdk #51: Attempting to boot Zephyr w/ MultiZone

AvocadoMatt, 2021-06-08:
I'm attempting to run this Zephyr [sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/hello_world) as a zone on the HiFive1 Rev B (FE310). I compile the sample with Zephyr's given SDK and then move it into its own directory inside of the MultiZone SDK. I run these commands to produce the hex (then fix the address as it points to 0x20010000) and lst files:
`toolchain-path/bin/riscv64-unknown-elf-objcopy -O ihex zephyr.elf zephyr-old.hex --gap-fill 0x00`
`toolchain-path/bin/riscv64-unknown-elf-objdump --all-headers --demangle --disassemble --file-headers --wide -D zephyr.elf > zephyr.lst`
`/root/workspace/multizone-riscv/riscv-gcc-toolchain/bin/riscv64-unknown-elf-objcopy --change-addresses 0x2800 zephyr-old.hex zephyr.hex`
I've updated the Makefile to appropriately use this new zone as such:
`$(MAKE) -C bsp/$(BOARD)/boot                                                   
 java -jar multizone.jar \                                                      
     --arch $(BOARD) \                                                            
     --config bsp/$(BOARD)/multizone.cfg \                                        
     --boot bsp/$(BOARD)/boot/boot.hex \                                          
     zephyr/zephyr.hex`
To receive this output after compiling:
`make -C bsp/FE310/boot clean
make[1]: Entering directory '/root/workspace/multizone-riscv/multizone-sdk/bsp/FE310/boot'
rm -f boot.o boot.hex boot.elf boot.lst boot.map
make[1]: Leaving directory '/root/workspace/multizone-riscv/multizone-sdk/bsp/FE310/boot'
rm -f multizone.hex
make -C bsp/FE310/boot
make[1]: Entering directory '/root/workspace/multizone-riscv/multizone-sdk/bsp/FE310/boot'
/root/workspace/multizone-riscv/riscv-gcc-toolchain/bin/riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -ffreestanding -Wall -x assembler-with-cpp -c -o boot.o boot.S
/root/workspace/multizone-riscv/riscv-gcc-toolchain/bin/riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -ffreestanding -Wall -T linker.lds -T ../memory.lds -nostdlib -Xlinker --gc-sections -Wl,-Map,boot.map -o boot.elf ./boot.o 
/root/workspace/multizone-riscv/riscv-gcc-toolchain/bin/riscv64-unknown-elf-objcopy -O ihex -j.boot boot.elf boot.hex
/root/workspace/multizone-riscv/riscv-gcc-toolchain/bin/riscv64-unknown-elf-objdump --source --all-headers --demangle --disassemble --line-numbers --reloc --wide boot.elf > boot.lst
make[1]: Leaving directory '/root/workspace/multizone-riscv/multizone-sdk/bsp/FE310/boot'
java -jar multizone.jar \
	--arch FE310 \
	--config bsp/FE310/multizone.cfg \
	--boot bsp/FE310/boot/boot.hex \
	zephyr/zephyr.hex
====================================================================
                   MultiZone Security Configurator                  
    Copyright 2020 Hex Five Security, Inc. - All Rights Reserved    
====================================================================
This version of MultiZone Security is meant for evaluation purposes 
only. As such, use of this software is governed by the Evaluation   
License. There may be other functional limitations as described in  
the evaluation SDK documentation. The commercial version of the     
software does not have these restrictions.                          
====================================================================
Kernel : section 0 address 0x20010000 size 0x0000170c 
Boot   : section 0 address 0x20012000 size 0x0000003c 
Zone 1 : section 0 address 0x20012800 size 0x00003f68 
Warning: zone 3 range 3 overlaps zone 2 range 4.

17:57:08 Build Finished (took 857ms)`
After running a make load to have the SEGGER J-Link flash the HiFive1 Rev B, I connect to the board via screen and receive no output after resetting the board.

MultiZone and Zephyr do work right out of the box in their respective directories, but I'm failing to get them to pair.

cgarlati, 2021-06-08:
Please post the content of your multizone.cfg file

cgarlati, 2021-06-08:
Also make sure you take a good look at the boot code example on branch https://github.com/hex-five/multizone-sdk/tree/example/boot-code

AvocadoMatt, 2021-06-09:
I left the multizone.cfg untouched, I saw the same results when joining the Zephyr sample with and without zones 2-4 in the multizone.jar.

> Copyright(C) 2020 Hex Five Security, Inc. - All Rights Reserved                
MultiZone reserved memory: 8K @0x20010000, 6K @0x80000000
Tick = 10 # ms 
Zone = 1                                                                         
    #irq  = 3 # DMA                                                               
    plic = 3 # UART                                                              
    base = 0x20012800; size =      30K; rwx = rx # FLASH                         
    base = 0x80001800; size =       4K; rwx = rw # RAM                           
    base = 0x10013000; size =    0x100; rwx = rw # UART                  
Zone = 2                                                                         
    #irq  = 20, 21, 22 # BTN0 BTN1 BTN2 (CLINT)                                   
    base = 0x2001A000; size =    8K; rwx = rx # FLASH                            
    base = 0x80002800; size =    2K; rwx = rw # RAM                              
    base = 0x10025000; size = 0x100; rwx = rw # PWM LED                          
    base = 0x10012000; size = 0x100; rwx = rw # GPIO                
Zone = 3                                                                         
    base = 0x2001C000; size =    8K; rwx = rx # FLASH                            
    base = 0x80003000; size =    2K; rwx = rw # RAM                              
    base = 0x10012000; size = 0x100; rwx = rw # GPIO                             
Zone = 4                                                                         
    base = 0x2001E000; size =    8K; rwx = rx # FLASH                            
    base = 0x80003800; size =    2K; rwx = rw # RAM

Looking at those links you posted

AvocadoMatt, 2021-06-09:
We're looking into using Zephyr as our RTOS for MultiZone, similar to how the [MultiZone IoT SDK](https://github.com/hex-five/multizone-iot-sdk/) can use a different RTOS such as FreeRTOS, however that SDK doesn't have a board support package for the HiFive1 Rev B (FE310). Is it possible to replace the RTOS in the TEE here?

cgarlati, 2021-06-09:
MultiZone protects the execution of **any binaries** without requiring modifications to the code. So it realy doesn't matter what's in the zones' HEX files as long as it pleases the CPU. It can be bare metal, assembly, C, ADA, FreeRTOS, Zephyr, Linux, hypervisors, mysterious proprietary / legacy stuff, etc. In fact, you can put total trash in the binary images and MultiZone will still safely run the remaining zones without interference.

If you want to run one or more Zephyr-based binaries, you have to map these binaries to zones and configure the security setting in the multizone.cfg file. The --boot option is for hardware setup / OEM low-level code that runs only once, at boot, and is then tossed upon starting the protected execution of the zones.

To your 2nd question: The reason why the [MultiZone IoT SDK](https://github.com/hex-five/multizone-iot-sdk) doesn't provide board support for the HiFive1 Rev B (FE310) is that this SoC has only 16KB of RAM, which makes it unsuitable for any practical application involving cryptograpy and/or TLS secure communications, like the MultiZone IoT SDK secure firmware.

STashakkori, 2021-06-11:
Hello Cesare and thank you for your help sir. We have spent a good bit of time trying to make this work with no success. We feel that we are close but it's possible that we are not. Is there a guide on how to configure Multizone to load a third-party OS like Zephyr or FreeRTOS? It would be a huge help to our team if we had this level of direction. The examples involve a baremetal application or the Multizone RTOS kernel and we have been able to do these successfully. However, our current step seems to need special knowledge we do not possess. Thanks again

STashakkori, 2021-07-24:
Pretty sure at this point that this is actually not a functionality of Multizone. Multizone is tightly coupled to run FreeRTOS applications or to run the Multizone "kernel". This is actually just an application that runs on a JVM in a baremetal environment. Multizone is not a hypervisor and hence does not support running of a kernel as the first thing the kernel will want to do is access hardware. The Multizone "TEE" is really just a way for baremetal applications to not interfere with each other. Seems rather minimal at that as well. In general our impression is that this technology is still under development. Hope this helps shed some light for anyone investigating this like we did. Will save them some time. Best, Sina

cgarlati, 2021-07-25:
Closing per request of the filer.

Note: some user comments moved to https://hex-five.com/faq/

closed 2021-07-25 by cgarlati

STashakkori, 2021-07-25:
We didn't close this issue actually. We discovered that Multizone doesn't do what it says it does and disclosed that information and they didn't like that we revealed the truth and they deleted my comment and closed this.

cgarlati, 2021-07-25:
@STashakkori
You have been temporarily banned from all Hex Five's repositories until you show more respect for the hard work of the many people involved in these projects.