hex-five/multizone-sdk #56: make load command not working

kavishnayeem, 2023-10-03:
when "make load " is executed there is an error called 
riscv-gnu-toolchain-20210618/bin/riscv64-unknown-elf-gdb: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by riscv-gnu-toolchain-20210618/bin/riscv64-unknown-elf-gdb)
 I am currently using virtualBox to emulate ubuntu 20 on my windows desktop. 
Any help is appreciated. Thank you

Hex-Five-Labs, 2023-10-03:
Kavish:

Library incompatibility issues across various releases / updates of Ubuntu are quite common and easy to fix.

What version(s) of libc.so.6 do you currently have in your system? 

Post the output of `apt policy libc6` in this thread and we'll take it from there.

kavishnayeem, 2023-10-05:
Thankfully I was able to solve that issue by upgrading my ubuntu to 22 version but there is still and error where it says :

Open On-Chip Debugger 0.11.0+dev-01860-g9aa8bab63-dirty (2021-08-08-01:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Error: no device found
Error: unable to open ftdi device with vid 15ba, pid 002a, description 'Olimex OpenOCD JTAG ARM-USB-TINY-H', serial '*' at bus location '*'
The target architecture is set to "riscv:rv32".
localhost:3333: Connection timed out.
"monitor" command not supported by this target.
"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.
"monitor" command not supported by this target.

Even  after running lsusb on my vm I can see that my olimex jtag is connected, but still I am encountering this error.
I hope this error can be resolved.
Thank you.

Hex-Five-Labs, 2023-10-05:
Kavish:

Good to hear you solved your library issue. Ubuntu 22 is a supported platform. OpenOCD/Olimex/FTDI work fine either natively or in a vm. MultiZone SDK tests include Ubuntu guest on Debian host - vmware workstation.

For your reference, this is the expected output of the `make load` script on Ubuntu 22.04 / vmware:

```
hexfive@vmware-pc:~/multizone-sdk$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

hexfive@vmware-pc:~/multizone-sdk$ lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

hexfive@vmware-pc:~/multizone-sdk$ make load
/home/hexfive/riscv-openocd-20210807/bin/openocd -f bsp/X300/openocd.cfg & \
/home/hexfive/riscv-gnu-toolchain-20210618/bin/riscv64-unknown-elf-gdb multizone.hex --batch -ex "set mem inaccessible-by-default off" -ex "set remotetimeout 240" -ex "set arch riscv:rv32" -ex "target extended-remote localhost:3333" -ex "monitor reset init" -ex "monitor flash protect 0 64 last off" -ex "load" -ex "monitor resume" -ex "monitor shutdown" -ex "quit"
Open On-Chip Debugger 0.11.0+dev-01860-g9aa8bab63-dirty (2021-08-08-01:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10300af9 (mfg: 0x57c (Hex Five Security Inc), part: 0x0300, ver: 0x1)
The target architecture is set to "riscv:rv32".
Info : datacount=1 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101105
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: riscv.cpu tap/device found: 0x10300af9 (mfg: 0x57c (Hex Five Security Inc), part: 0x0300, ver: 0x1)
Info : Found flash device 'micron n25q128' (ID 0x0018ba20)
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
0x08000080 in ?? ()
Info : JTAG tap: riscv.cpu tap/device found: 0x10300af9 (mfg: 0x57c (Hex Five Security Inc), part: 0x0300, ver: 0x1)
JTAG tap: riscv.cpu tap/device found: 0x10300af9 (mfg: 0x57c (Hex Five Security Inc), part: 0x0300, ver: 0x1)
cleared protection for sectors 64 through 255 on flash bank 0

cleared protection for sectors 64 through 255 on flash bank 0

Info : JTAG tap: riscv.cpu tap/device found: 0x10300af9 (mfg: 0x57c (Hex Five Security Inc), part: 0x0300, ver: 0x1)
Loading section .sec1, size 0x1a24 lma 0x20400000
Loading section .sec2, size 0x40 lma 0x20402000
Loading section .sec6, size 0x474 lma 0x20403000
Loading section .sec5, size 0xc34 lma 0x20404000
Loading section .sec4, size 0x87e lma 0x20406000
Loading section .sec3, size 0x4004 lma 0x20408000
Info : Padding image section 0 at 0x20401a24 with 1500 bytes
Info : Padding image section 1 at 0x20402040 with 4032 bytes
Info : Padding image section 2 at 0x20403474 with 2956 bytes
Info : Padding image section 3 at 0x20404c34 with 5068 bytes
Info : Padding image section 4 at 0x2040687e with 6018 bytes
Info : Disabling abstract command writes to CSRs.
Info : JTAG tap: riscv.cpu tap/device found: 0x10300af9 (mfg: 0x57c (Hex Five Security Inc), part: 0x0300, ver: 0x1)
Start address 0x20400000, load size 29582
Transfer rate: 3 KB/sec, 4226 bytes/write.
shutdown command invoked
shutdown command invoked
A debugging session is active.

	Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) [answered Y; input not from terminal]
Remote connection closed
```
A few troubleshooting steps:

- double check all MultiZone SDK Linux prerequisites in the README.md
- try native Linux - i.e. dual boot Windows / Ubuntu - as Windows drivers may very well be the cause of your problem
- make sure your Olimex head is properly wired to the ARTY board PMOD JD - double check all jumpers
- try a different or no usb hub if you are using one
- try a different Olimex head / usb cable

kavishnayeem, 2023-10-11:
I thank you for providing the expected output but now i am encountering a new error mentioned below:

/home/ramyab/riscv-openocd-20210807/bin/openocd -f bsp/X300/openocd.cfg & \
/home/ramyab/riscv-gnu-toolchain-20210618/bin/riscv64-unknown-elf-gdb multizone.hex --batch -ex "set mem inaccessible-by-default off" -ex "set remotetimeout 240" -ex "set arch riscv:rv32" -ex "target extended-remote localhost:3333" -ex "monitor reset init" -ex "monitor flash protect 0 64 last off" -ex "load" -ex "monitor resume" -ex "monitor shutdown" -ex "quit"
Open On-Chip Debugger 0.11.0+dev-01860-g9aa8bab63-dirty (2021-08-08-01:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
The target architecture is set to "riscv:rv32".
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
openocd: src/target/riscv/riscv-011.c:435: dbus_scan: Assertion `info->addrbits != 0' failed.
localhost:3333: Connection timed out.
"monitor" command not supported by this target.
"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.
"monitor" command not supported by this target.

The connection of JTAG is done by referring to the documentation provided by sifive and jtag producers (Olimex) But still we are getting this error . Any help is Appreciated. Thank you

Hex-Five-Labs, 2023-10-11:
Kavish, Ramya:

Go through this checklist and post the results of taking each step.
In particular, try native Linux or vmware Workstation vm on Linux host. 

Troubleshooting list:

- see https://github.com/hex-five/multizone-sdk/issues/48
- double check all MultiZone SDK Linux prerequisites in the README.md
- try native Linux - i.e. dual boot Windows / Ubuntu - as Windows drivers may very well be the cause of your problem
- make sure your Olimex head is properly wired to the ARTY board PMOD JD - double check all jumpers
- try a different or no usb hub if you are using one
- try a different Olimex head
- try a different usb cable
- try to power the ARTY from the barrel J13

kavishnayeem, 2023-10-16:
Thank you for your help I was able to resolve all the errors and execute every command. I want to explore more about multizones on RISC-V and will be working on it . I hope we stay connected in the future . Thank you for taking your time out for helping us . Appreciate it !

Hex-Five-Labs, 2023-10-31:
@kavishnayeem

Glad to know your team is up and running. For the benefit of future readers: can you please describe what turned out to be the cause of  your problem and how did you sove it?

closed 2023-10-31 by Hex-Five-Labs

kavishnayeem, 2023-10-31:
Thanks for checking on me , We were able to resolve the problem, it was  the jtag connections for our particular board . We followed https://static.dev.sifive.com/SiFive-E310-arty-gettingstarted-v1.0.6.pdf for correct connections . Now our team is working on applying machine learning in multizones where we divide the machine learning tasks and pass it to different zones for computation and predict the result in zone1. Lets hope for the best. 
If there will be any issues in the future I will surely contact the Hex-five team
Thank you Hex-Five for helping us.