hex-five/multizone-sdk #60: Details on boards that supports multizone

RamyaBoggaram, 2023-10-31:
To continue we started with the official website https://github.com/hex-five/multizone-sdk on our virtual box with ubuntu machine and we are trying to connect "Arty A7 100T" board. Now we are working on the arty a7 board and now we are curious to explore more on the boards that could help us in Achieving our goal. Can you let me know on the details of other boards that we could try to work on.

Hex-Five-Labs, 2023-10-31:
List of supported hardware development kits as of November 2023:

**RISC-V**

- [Digilent Arty A7 Development Board (Xilinx Artix-7 FPGA)](https://digilent.com/shop/arty-a7-artix-7-fpga-development-board/)
- [Andes Corvette-F1 R1.0 (Xilinx Artix-7 FPGA)](http://www.andestech.com/en/products-solutions/andeshape-platforms/corvette-f1-r1/)
- [Microchip Icicle Kit (PolarFire SoC)](https://www.microsemi.com/existing-parts/parts/152514)
- [SiFive HiFive1 Rev B (Freedom E310 SoC)](https://www.sifive.com/boards/hifive1-rev-b)
- [SiFive Unleashed (Freedom U540 SoC)](https://www.sifive.com/boards/hifive-unleashed)

**Arm Cortex-M**

- [GigaDevice GD32307C-EVAL evaluation kit (GD32F207 Cortex-M4 @120MHz)](https://www.gigadevice.com/products/microcontrollers/gd32-development-tools/gd32-evaluation-boards/)
- [Microchip SMART SAM E70 Xplained evaluation kit (ATSAME70Q21 Cortex-M7 @300MHz)](http://www.microchip.com/developmenttools/productdetails/atsame70-xpld)
- [NXP MIMXRT1020-EVK (i.MX RT1020 Cortex-M7 @500MHz)](https://www.nxp.com/design/development-boards/i.mx-evaluation-and-development-boards/i.mx-rt1020-evaluation-kit:MIMXRT1020-EVK) 
- [Renesas EK-RA6M3 (R7FA6M3AH3CFC Cortex-M4 @120MHz)](https://www.renesas.com/us/en/products/software-tools/boards-and-kits/eval-kits/ek-ra6m3.html)
- [ST NUCLEO-F767ZI (STM32F767ZI Cortex-M7 @216MHz)](https://www.st.com/en/evaluation-tools/nucleo-f767zi.html)
- [NXP i.MX 8QuadMax](https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/i-mx-8quadmax-multisensory-enablement-kit-mek:MCIMX8QM-CPU)

RamyaBoggaram, 2023-11-01:
Thanks for your reply!!

closed 2023-12-13 by Hex-Five-Labs